micropython/micropython-esp32-ulp #10: using not case-sensitive symbols?

ThomasWaldmann, 2018-03-11:
Usually I am all for case-sensitivity, because it avoids a lot of issues.

But considering the symbol names from espressif have a lot of SYMBOLS_LIKE_THAT, I find that rather ugly, so I am considering just ignoring case, so typing the same stuff in lowercase does also match.

Opinions?

igrr, 2018-03-12:
Do you mean symbols in the "symbols in object file" sense or as in identifiers? If we have upper case symbols in object files somewhere, that's likely a but which we should fix. If this is about assembly tokens, then not making them case sensitive makes sense, as this is how binutils works as well. FWIW, [ULP examples](https://github.com/espressif/esp-idf/blob/master/examples/system/ulp/main/ulp/pulse_cnt.S) use lower case, except for preprocessor macros.

ThomasWaldmann, 2018-03-12:
I meant stuff in assembly source. Instructions and also long-named constants for specific SoC register addresses / memory areas. The latter is somehow convention to be uppercase, but still, typing / reading uppercase is not nice.

I had mostly looked at `ulp.h` at the time I wrote that. I am using the tech ref and other non-deprecated docs now.

Yes, that example already looks much better.

ThomasWaldmann, 2018-03-13:
ok, so does that look ok?:

- mnemonics: ulp instructions: not case-sensitive
- mnemonics: assembler directives: not case-sensitive
- operands: registers: not case-sensitive
- operands: conditions (eq, ov, ge, ...): not case-sensitive
- labels / symbols: case-sensitive

igrr, 2018-03-13:
Yes, this looks reasonable!

ThomasWaldmann, 2018-03-19:
guess this is how it is currently implemented, closing.

closed 2018-03-19 by ThomasWaldmann