micropython/micropython-esp32-ulp #12: add useful examples

ThomasWaldmann, 2018-03-12:
when the assembler starts working good enough and micropython has added the ulp api functions, we'll need some example code, like:

- py: assemble, load and run some ulp code
- asm: ulp blink
- py/asm: give data to / fetch data from ulp

mattytrentini, 2018-03-20:
I'm keen to see an I2C example; specifically I have a long-term goal to use the ULP to periodically query an I2C temperature sensor, store it in RTC RAM and, when the RTC RAM is close to exhaustion, wake up the main processor to store the data in secondary storage or offload it over wifi.

I'll admit it's a large chunk of work at this stage! 

I'll try running the current examples first but will work toward this goal...

ThomasWaldmann, 2018-03-20:
@mattytrentini that could even maybe become a little framework.

mattytrentini, 2018-04-21:
Is there a simple 'hello world' example?

In particular, it'd be great to see a GPIO toggled at a slow rate where an LED could be connected and it would be possible to see it flash on and off.

ThomasWaldmann, 2018-04-22:
i could not yet test any ulp binary as the load/run functions are not merged yet into micropython (nor do i have a working setup to compile my own micropython yet).

thus, no practically usable ulp example code from me yet, sorry.

mattytrentini, 2018-04-23:
Thanks for the update @ThomasWaldmann. I _have_ built MicroPython with the ULP methods; I can upload the binary somewhere so you can use it on your ESP32 if you like? 

I spent some time yesterday learning more about ULP assembly programming with the goal to come up with that flashing LED example. I just didn't have enough time to progress very far! One issue I ran into was that the Espressif binutils use a C preprocessor before compiling and they provide [convenience macros](http://esp-idf.readthedocs.io/en/latest/api-guides/ulp_instruction_set.html#convenience-macros-for-peripheral-registers-access) to help manipulate registers, including GPIO's. You'll see most ULP examples `#include` `"soc/rtc_io_reg.h"` and `"soc/soc_ulp.h"` etc which is where the macros are defined. We may need some sort of similar (maybe the same?) facility to make register access easier. In the meantime I'll unpack the macros and do whatever they're doing...

mattytrentini, 2018-04-23:
Use `esptool` to write this [MicroPython firmware with ULP features](https://github.com/ThomasWaldmann/py-esp32-ulp/files/1938224/esp32_micropython_ulp.zip)

ThomasWaldmann, 2018-05-02:
@mattytrentini thanks! i didn't have time when you posted it and a few days later it now got finally merged into micropython, so the ulp load/run stuff is contained in the daily builds after 2018-05-01.

I tried them, it works. Have just added a first practically tested example to examples/ directory.

mattytrentini, 2018-05-04:
I've just tried your `counter.py` example on the daily MicroPython ESP32 firmware - it all works! Nice job @ThomasWaldmann :+1:

mattytrentini, 2018-06-11:
Heya Thomas,

I was trying to modify your counter example to toggle a GPIO line but came a little unstuck. Here's some code:

```
data:       .long 0

entry:      move r3, data
            ld r2, r3, 0

            and r1, r2, 255               # AND r2 with 0xff to set the GPIO register appropriately
            reg_wr 0x3ff48400, 26, 26, 1  # Write to GPIO 2
            halt
```

I was just trying to _set_ GPIO2 (I know I'm going to need to branch to toggle the register because of #18). However, after I run this code then try to read the GPIO value from MicroPython (`Pin(2).value()`) it would always be unaffected by the ULP code. Multimeter probing also showed the pin didn't appear to toggle.

I tried a couple of alternatives: using 0 instead of 1 in reg_wr, setting Pin 2 as an output pin before executing the ULP code, used a different GPIO. But it didn't seem to work. 

I double-checked the `reg_wr` values - these values are deduced by expanding the macros supplied in the ESP-IDF:

```
// GPIO2 LED ON
WRITE_RTC_REG(RTC_GPIO_OUT_REG, RTC_GPIO_OUT_DATA_S + 12, 1, 1)

soc_ulp.h
    // Write immediate value into rtc_reg[low_bit + bit_width - 1 : low_bit], bit_width <= 8
    #define WRITE_RTC_REG(rtc_reg, low_bit, bit_width, value) \
    REG_WR (((rtc_reg) - DR_REG_RTCCNTL_BASE) / 4), ((low_bit) + (bit_width) - 1), (low_bit), ((value) & 0xff)

soc.h
    #define DR_REG_RTCCNTL_BASE                     0x3ff48000
    #define DR_REG_RTCIO_BASE                       0x3ff48400

rtc_io_reg.h
    #define RTC_GPIO_OUT_REG          (DR_REG_RTCIO_BASE + 0x0)
    #define RTC_GPIO_OUT_DATA_S  14
```

So I'm just wondering if you've tried setting a GPIO from ULP yourself? I'm not sure if the register requires initialization (hold value? output pin?) but _something_ isn't working. Any ideas? :)

ThomasWaldmann, 2018-06-11:
Didn't try much practical ULP programming yet, but yes, the addrs are different in ULP coproc address space.

mrwhy-orig, 2019-02-14:
I'm fairly new to µC and µPython. That said, I think it would be much easier to get used to the ULP and your module if you could supply an example how to achieve the outcome of the Espressif example:
[](https://github.com/espressif/esp-idf/tree/a6e4e895920d2bae3cbe34c61b89ee42770d5fa8/examples/system/ulp)

ThomasWaldmann, 2019-02-14:
You'ld need to adapt that a bit. Our code does not process `#include`. Not sure about the other meta opcodes used there. Just try it! :)

If you get it working, make a PR.