micropython/micropython-esp32-ulp #13: add CI?

igrr, 2018-03-12:
I can submit a PR to run some code ULP code through py-esp32-ulp and through binutils-esp32ulp, and compare the generated code, if you think this is useful.

ThomasWaldmann, 2018-03-12:
Ah, yes, that would be very useful! For travis-ci or do you have sth else in mind?

(but note that the py-esp32-ulp assembler is not quite ready *yet*, but hopefully soon)

ThomasWaldmann, 2018-03-13:
`.S --> .ulp` works now (the latter being the format with header expected by the loader).

Totally untested, but looking reasonable.

I made up that .ulp extension, what's the usual one?

igrr, 2018-03-13:
Yes, i was thinking about adding a travis-ci script.

It's not going to fully work initially also because the binutils doesn't support SoC register addresses (like 0x3ff4-something) as arguments of reg_rd and reg_wr instructions. But that's something that's easy to add on binutils side of things, and looks like a nice feature.

The idea is to commit some simple tests first (which will pass in the current state of things) and add more tests as the features get added and compatibility improves.

Regarding file extension: IDF build process generates ulp binaries with .bin extension, which then get embedded into the firmware. The whole process is mostly invisible to the users, so i didn't put much thought into the extensions. ULP is fine although i have to say that it may not be future proof. Future chips might be using different architecture for the ULP, so perhaps it would be better to indicate that somehow (.ulp32? .esp32ulp? I don't know if extensions need to be 3 characters or not...)

ThomasWaldmann, 2018-03-13:
Ideally it should be something the micropython filesystem can accept. Have to try that later.

CI plans sound good! :)

ThomasWaldmann, 2018-03-13:
thanks for the PR, we have CI now! \o/

one small issue: the first 5000 lines of output are rather uninteresting (if binutils-esp32 and micropython build as usual), can we get them collapsed?

igrr, 2018-03-13:
Sure, will wrap them into folds.

ThomasWaldmann, 2018-03-13:
also please split the pure python stuff from the binutils stuff (creating 2 separate scripts to run them).

binutils-esp32 does not compile on my machine (bug reported), so I don't have that locally, so the one script does not work any more.

ThomasWaldmann, 2018-03-19:
fixed by #23 and #24.

closed 2018-03-19 by ThomasWaldmann