micropython/micropython-esp32-ulp #18: reg_wr, i2c_wr - writing only immediate values?

ThomasWaldmann, 2018-03-13:
Hmm, AFAICS these 2 instructions only write constant data (given as immediate data, stored inside the instruction) to registers or i2c.

So, how would one write variable data?

igrr, 2018-03-13:
As far as i know, there are two options:
- switch/case constructs to write one of predefined values, and
- self-modifying code

ThomasWaldmann, 2018-03-13:
Huh, that is dirty.

Was that design by mistake? It could also have used a register for the data maybe?

(what I mean is an arbitrary byte value, like e.g. a brightness value for something, 0..255)

igrr, 2018-03-13:
Well, the whole ULP thing initially was a very-very primitive FSM designed to poll ADCs and write values to memory. It got some features bolted on to it in the very final days of the design process: arithmetic instructions, loops, jumps. But there was not enough time to make all the things generic enough (like for example reg_wr), and verify the changes. So yes, it would definitely have been great if we could reg_wr from register, but that late feature just didn't make it.

ThomasWaldmann, 2018-03-13:
Ah, that explains it.

Luckily these unused/reserved bits in the instructions are usually set to 0, so a later, compatible fix/extensions looks possible, even without using the free opcode numbers.

closed 2018-03-13 by ThomasWaldmann

mattytrentini, 2018-05-20:
I just stumbled over this today; was trying to create an example where I would use the ULP to control GPIO lines.  Note it was also [discussed on the Espressif  forum](https://www.esp32.com/viewtopic.php?t=3707).

In my case I can easily use a branch since I was just setting 1 or 0 but this could be a serious limitation...