micropython/micropython-esp32-ulp #19: macros?

ThomasWaldmann, 2018-03-13:


mattytrentini, 2018-05-20:
Yes, I think we need a replacement for these. A few implementation options off the top of my head:

1. Actually run the source through a (probably rudimentary) C preprocessor 
2. Extend the parser to know about those specific macros and replace them
3. Extend the opcodes to introduce new higher-level commands that don't actually exist in the ulp

I think 1) will be too heavy-weight. There are even existing C preprocessors implemented in Python that could be used but they are all complex and memory intensive. Suspect it'll be _very_ hard to work within the memory constraints of a microcontroller.

2 could work. Perhaps makes sense to explicitly call it before `src_to_binary`:

``` binary = src_to_binary(replace_macros(source)) ```

`replace_macros` would drop the includes and carefully replace the macros defined by Espressif. 

The attractive part of 3 is that it could make the code quite elegant. But the commands would be specific to py-esp32-ulp and it could be misleading to users; it wouldn't be clear how they mapped to ULP instructions. 

Worth noting that 1 & 2 have a huge benefit of being compatible with existing ULP code.

I think I'm leaning toward option 2...

mattytrentini, 2018-05-20:
I'm also a little confused with the implementation of the Espressif macros. It appears like the address passed in to the macro should be a 32bit address but by the time the macro is expanded the value is offset and divided by 4:

```
// GPIO2 LED ON
WRITE_RTC_REG(RTC_GPIO_OUT_REG, RTC_GPIO_OUT_DATA_S + 12, 1, 1)
```

Expands with (from [soc_ulp.h](https://github.com/espressif/esp-idf/blob/master/components/soc/esp32/include/soc/soc_ulp.h)):

```
#define WRITE_RTC_REG(rtc_reg, low_bit, bit_width, value) \
    REG_WR (((rtc_reg) - DR_REG_RTCCNTL_BASE) / 4), ((low_bit) + (bit_width) - 1), (low_bit), ((value) & 0xff)
```

to:
```
REG_WR ((0x3ff48400 - 0x3ff48000) / 4, ((14 + 12), (14 + 12), (1 & 0xff)))
```

It just looks...wrong. Your [compatibility test](https://github.com/ThomasWaldmann/py-esp32-ulp/blob/master/tests/compat/io.S) takes the raw `0x3ff48000` address. 

Am I missing something?

ThomasWaldmann, 2018-05-20:
I think option 1) is out of question due to the reasons you already mentioned.

2) [good compatibility] and 3) [ease of implementation] seem to be good options

BTW, I noticed that this ticket is about 2 different kinds of macros:
- the topmost comment is about assembler macros (== look like a assembler directive)
- the 2nd comment is about C-preprocessor-like macros (== look like a C function call)

lazyoldjack, 2020-11-23:
As a potential user, option 2 looks the best to me, though I have to admit to currently knowing **very** little about these macros, except for the fact that I need to use them in my ULP code to read and write pins. It's a shame this project is so quiet because I'm sure that there are many  ESP32 users who need the simplicity of MicroPython for a load of complex stuff but still need the ease of just including a snippet of ULP code in their project to save on battery power.

If there is anything I can do to help advance this issue, please let me know, though I fear I won't be much use except for testing.

ThomasWaldmann, 2020-11-23:
Yeah, not many contributors here (not even me right now). Maybe it needs more advertisement in the esp32 community.

Advancing this stuff == making a pull request.

lazyoldjack, 2020-12-28:
OK, so Matty wrote the code for the ULP WAKE instruction in MicroPython, so that's a big improvement to ULP use imho because MP can be restarted based on an ULP algorithm. I (and others) have tested it and it works with py-esp32-ulp. Hurrah. 🥳 

I think the difficulty of performing IO etc. without the macros will have put off many of the potential users of ULP code in MP, so I am going to look to see if I can understand what's involved in adding them, but as Matty will tell you, I didn't get very far with trying to work out the code for the WAKE instruction so don't hold your breath! 😸 

Thanks again for making such a great tool. I think when the macros work, there might be a possibility of getting Andreas Spiess to create a feature video on py-esp32-ulp, especially if I can furnish him with some working examples and that is likely to generate a lot more interest.

lazyoldjack, 2020-12-29:
... and another excellent use of the ULP is if you want to perform some bit bashing, because it is far less subject to timing interference then the main cores that are under the influence of RTOS. Here is a colour coded persistence trace of an RTCIO pin being bashed in the ULP; I never saw one pulse out of place.
[image]
I guess that's why some folks have tried to use it for audio.

ThomasWaldmann, 2021-08-09:
Fixed by #43.

closed 2021-08-09 by ThomasWaldmann