micropython/micropython-esp32-ulp #42: Trouble running example

errorfixrepeat, 2019-03-24:
Firstly, thanks for contributing this to Micropython. It seems very promising.

Unfortunately, I am unable to run the example code. I get the following error:
`Traceback (most recent call last):
  File "<stdin>", line 37, in <module>
OSError: 260`

This occurs  at `ulp.load_binary(load_addr, binary)`. Googling the error it seems to be an issue with the binary. The binary generated is:
`>>> print(binary)
b'ulp\x00\x0c\x00\x18\x00\x00\x00\x00\x00\x00\x00\x00\x00\x03\x00\x80r\x0e\x00\x00\xd0\x1a\x00\x00r\x0e\x00\x00h\x00\x00\x00\xb0'`

ThomasWaldmann, 2019-03-24:
Well, last time I tried, it worked for me.

At a quick glance, the binary looks legit, what did you use as load address?

Also, the MicroPython version you used and the source code is missing to reproduce the issue.

You could also compare the binary to the output of the assembler from espressif.

errorfixrepeat, 2019-03-24:
My apologies, here's some more info:

I'm using the current version from micropython.org:
`MicroPython v1.10-230-ge0c6dfe90 on 2019-03-24; ESP32 module with ESP32`

Code used:
```
"""
Very basic example showing data exchange main CPU <--> ULP coprocessor.

To show that the ULP is doing something, it just increments the value <data>.
It does that once per ulp timer wakeup (and then the ULP halts until it gets
waked up via the timer again).

The timer is set to a rather long period, so you can watch the data value
incrementing (see loop at the end).
"""

from esp32 import ULP
from machine import mem32

from esp32_ulp.__main__ import src_to_binary

source = """\
data:       .long 0

entry:      move r3, data    # load address of data into r3
            ld r2, r3, 0     # load data contents ([r3+0]) into r2
            add r2, r2, 1    # increment r2
            st r2, r3, 0     # store r2 contents into data ([r3+0])

            halt             # halt ULP co-prozessor (until it gets waked up again)
"""

binary = src_to_binary(source)

load_addr, entry_addr = 0, 4

ULP_MEM_BASE = 0x50000000
ULP_DATA_MASK = 0xffff  # ULP data is only in lower 16 bits

ulp = ULP()
ulp.set_wakeup_period(0, 50000)  # use timer0, wakeup after 50.000 cycles
ulp.load_binary(load_addr, binary)

mem32[ULP_MEM_BASE + load_addr] = 0x1000
ulp.run(entry_addr)

while True:
    print(hex(mem32[ULP_MEM_BASE + load_addr] & ULP_DATA_MASK))

```

Going by this I believe the load address is `0`

ThomasWaldmann, 2019-03-24:
Looks like my example code. You didn't modify anything?

errorfixrepeat, 2019-03-26:
It is. Trying to walk before I run!

errorfixrepeat, 2019-03-26:
I went back to the stable version of Micropython `esp32spiram-20190125-v1.10.bin`, and all is good. Apologies for wasting your time.

ThomasWaldmann, 2019-03-26:
Oh? So they have a regression in daily build? Then you should file a bug on the mp issue tracker.

bruno963852, 2019-09-25:
I compiled from source, both HEAD and v1.11 and it has the same problem...

ThomasWaldmann, 2019-09-26:
@bruno963852 ok, so there seem to be either a incompatible change or a regression in micropython between 1.10 and 1.11.

Can you check the change log and if there is nothing noted, open an issue in the micropython issue tracker?

ThomasWaldmann, 2019-09-29:
Working on this.

Confirmed: counter.py works on micropython 1.10 release.

ThomasWaldmann, 2019-09-29:
Analyzing the generated binary:
```
header:
ulp\x00  magic
\x0c\x00 load addr text
\x18\x00 size text
\x00\x00 size data
\x00\x00 size bss

data:
\x00\x00\x00\x00 long 0 (counter value)

entry (code):
\x03\x00\x80r
\x0e\x00\x00\xd0
\x1a\x00\x00r
\x0e\x00\x00h
\x00\x00\x00\xb0
```

ThomasWaldmann, 2019-09-29:
Related error codes, got them from esp-idf components/esp_common/src/esp_err_to_name.c:
```
    ERR_TBL_IT(ESP_OK),                             /*     0 esp_err_t value indicating success (no error) */
    ERR_TBL_IT(ESP_ERR_INVALID_SIZE),               /*   260 0x104 Invalid size */
    ERR_TBL_IT(ESP_ERR_INVALID_ARG),                /*   258 0x102 Invalid argument */
    ERR_TBL_IT(ESP_ERR_NOT_SUPPORTED),              /*   262 0x106 Operation or feature not supported */
```

So, 260 is ESP_ERR_INVALID_SIZE.

Current esp-idf code:
```
esp_err_t ulp_load_binary(uint32_t load_addr, const uint8_t* program_binary, size_t program_size)
{
    size_t program_size_bytes = program_size * sizeof(uint32_t);
    size_t load_addr_bytes = load_addr * sizeof(uint32_t);

    if (program_size_bytes < sizeof(ulp_binary_header_t)) {
        return ESP_ERR_INVALID_SIZE;
    }
    if (load_addr_bytes > CONFIG_ESP32_ULP_COPROC_RESERVE_MEM) {
        return ESP_ERR_INVALID_ARG;
    }
    if (load_addr_bytes + program_size_bytes > CONFIG_ESP32_ULP_COPROC_RESERVE_MEM) {
        return ESP_ERR_INVALID_SIZE;      # XXX is this the problem??? XXX
    }

    // Make a copy of a header in case program_binary isn't aligned
    ulp_binary_header_t header;
    memcpy(&header, program_binary, sizeof(header));

    if (header.magic != ULP_BINARY_MAGIC_ESP32) {
        return ESP_ERR_NOT_SUPPORTED;
    }

    size_t total_size = (size_t) header.text_offset + (size_t) header.text_size +
            (size_t) header.data_size;

    ESP_LOGD(TAG, "program_size_bytes: %d total_size: %d offset: %d .text: %d, .data: %d, .bss: %d",
            program_size_bytes, total_size, header.text_offset,
            header.text_size, header.data_size, header.bss_size);

    if (total_size != program_size_bytes) {
        return ESP_ERR_INVALID_SIZE;      # XXX or is it that? XXX
    }

    size_t text_data_size = header.text_size + header.data_size;
    uint8_t* base = (uint8_t*) RTC_SLOW_MEM;

    memcpy(base + load_addr_bytes, program_binary + header.text_offset, text_data_size);
    memset(base + load_addr_bytes + text_data_size, 0, header.bss_size);

    return ESP_OK;
}
```

ThomasWaldmann, 2019-09-29:
I tried to load a binary with invalid magic, it still errors out with 260.

That means it does not reach the code that checks the binary, so it likely fails at the first check:

```
    if (load_addr_bytes + program_size_bytes > CONFIG_ESP32_ULP_COPROC_RESERVE_MEM) {
        return ESP_ERR_INVALID_SIZE;
```

So, was the breaking change that micropython > 1.10 was compiled with CONFIG_ESP32_ULP_COPROC_RESERVE_MEM == 0 or so?

ThomasWaldmann, 2019-09-29:
Reproduction code:

```
from esp32 import ULP

bin = b'ulp\x00\x0c\x00\x18\x00\x00\x00\x00\x00\x00\x00\x00\x00\x03\x00\x80r\x0e\x00\x00\xd0\x1a\x00\x00r\x0e\x00\x00h\x00\x00\x00\xb0'

assert len(bin) == 36

ulp = ULP()
ulp.load_binary(0, bin)

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: 260
```

ThomasWaldmann, 2019-09-29:
Filed an issue there: https://github.com/micropython/micropython/issues/5159

ThomasWaldmann, 2021-08-09:
upstream bug was fixed / closed, closing here, too.

closed 2021-08-09 by ThomasWaldmann