micropython/micropython-esp32-ulp #61: find a new home for this project

ThomasWaldmann, 2021-11-21:
while i did create this project and did a lot of the ground work, i didn't work too much on it recently nor did i practically use it too much in practice.

also, i generally think that a project that's not just for personal use but somehow generally useful should rather live at some "neutral" place (like in a github organisation) than in someone's personal github account.

@wnienhaus did a lot of work recently, so this is now much more practically usable than it was before, so i guess this is now ready to be used and maintained by a community of interested people.

so, let's collect ideas about where this should live in future (github orgs)...

ThomasWaldmann, 2021-11-21:
https://github.com/micropython

https://github.com/espressif

kjm1102, 2022-02-05:
I can't get it to run
`Traceback (most recent call last):
  File "<stdin>", line 4, in <module>
ImportError: no module named 'esp32_ulp'`

wnienhaus, 2022-02-05:
That is surprising. Could you paste here what command you used and which port of MicroPython you tried it with?

Some things to check/try:
* In case you used the unix port and cloned this repo somewhere, make sure you are in the root directory of this repository (not in the esp32_ulp subdirectory) when trying `import esp32_ulp`.
* If you are using the esp32 port and used upip to install the micropython-py-esp32-ulp module, then upip by usually installs this into the `/lib` directory. Ensure that `/lib` is part of your path, i.e. `import sys; print(sys.path)`

ThomasWaldmann, 2022-02-05:
@kjm1102 @wnienhaus please move this to a new ticket (see issue title).

kjm1102, 2022-02-06:
tnx, wnienhaus got it working.  I would if I knew how to start a new ticket Thomas.

ThomasWaldmann, 2022-02-06:
There's a "new issue" green button at the top right of the page.

ThomasWaldmann, 2022-02-24:
@MattTrentini said: https://twitter.com/matt_trentini/status/1496827362830602240

mattytrentini, 2022-02-26:
I brought this up with Damien during the meetup; he seemed open to the idea of moving `py-esp32-ulp` to the [MicroPython organization](https://github.com/micropython) and giving you full access. If that was something you wanted...

wnienhaus, 2022-02-26:
@ThomasWaldmann @mattytrentini I think that sounds great. It would give the project a life of its own without being directly connected to TW and me. I'm still happy to help maintain it - like fixing issues, improving docs, encouraging/reviewing PRs, etc

ThomasWaldmann, 2022-02-26:
OK, so we'll move this to MicroPython organization. \o/

ThomasWaldmann, 2022-02-26:
I tried to transfer, but getting:
```
You don’t have the permission to create public repositories on micropython
```

@dpgeorge I'll just give you full access here, so you can initiate the transfer of this repo/issues/etc. to micropython organization.

dpgeorge, 2022-02-28:
According to github docs, only @ThomasWaldmann can transfer this repo.  To do that you need to be a member of the micropython org and have sufficient rights.

@ThomasWaldmann let me know when you're online and we can run through the process together (using the micropython slack channel?)

ThomasWaldmann, 2022-02-28:
I'm not on slack, but I'll try to transfer again now...

ThomasWaldmann, 2022-02-28:
```
You don’t have the permission to create public repositories on micropython
```

dpgeorge, 2022-03-01:
Can you DM me on slack?

dpgeorge, 2022-03-01:
Oh, sorry, I see you said you are *not* on slack

dpgeorge, 2022-03-01:
I sent you an invite to the micropython org.  Once you accept I need to give you sufficient access to transfer the repo.

ThomasWaldmann, 2022-03-01:
Accepted invite, it said "you're a member now...", but it still does not let me transfer. Same error msg as above.

ThomasWaldmann, 2022-03-01:
Do your org permissions let members create repos?

dpgeorge, 2022-03-01:
I added you to a team that should have permissions to push a repo.

ThomasWaldmann, 2022-03-01:
Ah, you need to adjust it first, sorry, overread that. OK, trying again...

ThomasWaldmann, 2022-03-01:
Still same error msg.

dpgeorge, 2022-03-01:
Ok, let me try something else.  I'll let you know when to try again.

ThomasWaldmann, 2022-03-01:
BTW, thanks for micropython. Loved how easy it is nowadays to use microcontrollers thanks to it!

dpgeorge, 2022-03-01:
Ok, please try again.

ThomasWaldmann, 2022-03-01:
Looks like it worked! :-) Great new home!

dpgeorge, 2022-03-01:
Looks like that worked!

Did you want to rename the repo to match other ones there, like `micropython-esp32-ulp`??  I don't know the history of this repo so I'm probably not in a good position to suggest a name change, but if you wanted to do that, now would be the best time.

dpgeorge, 2022-03-01:
Does @wnienhaus want push access to this repo?

ThomasWaldmann, 2022-03-01:
https://pypi.org/project/micropython-py-esp32-ulp/ this is our pypi package.

Options for repo name:

- micropython-py-esp32-ulp (same as on pypi)
- micropython-esp32-ulp (shorter)

Now is a good time to rename as it moved anyway. When chosing the 2nd option, we would need to rename more stuff, like the pypi package and guess also the package name.

ThomasWaldmann, 2022-03-01:
Please give @wnienhaus full access here.

ThomasWaldmann, 2022-03-01:
@dpgeorge let's rename to 1st option now (less stuff to adjust).

In case @wnienhaus and me decide differently later, we'ld need to change quite some more stuff anyway.

dpgeorge, 2022-03-01:
> let's rename to 1st option now (less stuff to adjust).

That sounds good.

You don't need to rename it, there are other repos in this org which are not prefixed with `micropython-`.  The `micropython-` prefix means it's intended for use with MicroPython only, which is probably true for this repo.  And renaming it to match the PyPI package makes sense as well.

So rename if you feel it makes sense.  If you don't have permissions let me know and I'll do it.

ThomasWaldmann, 2022-03-01:
Guess you need to do that, i do not have "settings" here.

dpgeorge, 2022-03-01:
I renamed it.

ThomasWaldmann, 2022-03-01:
Yup, that worked, I already forked a private clone from it from the new location.

I'll advertise the new location on twitter / mastodon. Can you write something on micropython's channels?

ThomasWaldmann, 2022-03-01:
TODO: change project URL in `setup.py` / pypi metadata and maybe other docs.

ThomasWaldmann, 2022-03-01:
@dpgeorge gtg now. thanks for helping!

wnienhaus, 2022-03-13:
Great work moving the repo to its new home.

I quite like the shorter "micropython-esp32-ulp". In the long name, the "py" feels redundant. The shorter name would also align nicely with the "esp32_ulp" module name inside the repo, i.e. micropython-esp32-ulp is the "esp32_ulp" module for micropython.

I made a PR (#68) with the adaptation to the shorter name, to "get a feel" for the different name. We can use it to decide which looks/feels better. If we choose the existing name, then PR #67 is good to merge as is, and we simply throw away my PR.

If we do go the route of the changed name, I think the only thing that's needed more, is to create a pypi package with the new name on both the main pypi and the test pypi. I can do that, but I have no access to the Github repo settings, so updating the API keys would need to be done by someone still (wondering though, if I should have access to the settings).

Let me know which name feels nicer to go forward with, and I can create the pypi repos if necessary.

ThomasWaldmann, 2022-03-14:
I also do not have access to the repo settings here. So guess you'ld need to do that together with damien.

I do not have strong opinions about the name, your approach is maybe slightly nicer, mine was a bit easier to do. But if you like to change it, just do it together with damien and create a new release so everything is consistent and uptodate.

mattytrentini, 2022-03-17:
So is the intent to use `micropython-esp32-ulp`? Are we waiting for @dpgeorge to make that change?

I want to announce the project (forum, slack, twitter...) but would prefer to do so _after_ a name change 😜

wnienhaus, 2022-03-18:
I would propose the name change, yes. But I'm happy to hear more opinions on this too.

But practically, I would need @dpgeorge to either give me (and TW) full access to this repo, or to change the name for us and then (potentially) update the API tokens for pypi once I created the new projects there.

dpgeorge, 2022-03-20:
I have now made @ThomasWaldmann and @wnienhaus administrators of this repo.  You should now be able to do anything...!

I'm happy with the proposed change to `micropython-esp32-ulp`.  If anything, it's shorter and simpler (that with the extra `-py`).  But no strong feelings (I have very little experience with this project).

wnienhaus, 2022-03-26:
Thank you @dpgeorge for the admin access to this repo.

I have now updated the repo to its new name and merged the PR, which updates source and documentation to use the new name. I released this as version `1.2.0` to PyPI and documented this (non-functional) change in the `CHANGES` file.

(As part of this, I have updated the API tokens for publishing to test-pypi and pypi. I have also invited @ThomasWaldmann to those PyPi projects with "owner" rights, so I am not a single point of failure here. Perhaps I should invite you @dpgeorge too?)

@mattytrentini you may now announce the project. Thank you for helping to spread the news!

dpgeorge, 2022-03-28:
Good work!

I think this issue can be closed now.

closed 2022-03-28 by ThomasWaldmann

kjm1102, 2022-04-25:
Glad this project found a new home but, like the first time I tried to use it, I'm still struggling.
`>>> upip.install('micropython-py-esp32-ulp')
Installing to: /lib/
Warning: micropython.org SSL certificate is not validated
Installing micropython-py-esp32-ulp 1.1.0 from https://files.pythonhosted.org/packages/b6/59/90b952a291882ad595c3d06bd0bdd0de582017e02b18293136430ebf3dd7/micropython-py-esp32-ulp-1.1.0.tar.gz
>>> import counter
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: no module named 'counter'`

Also I can't work out how to access docs/index.rst.