micropython/micropython-esp32-ulp #69: LED is not blinking

technical-publishing, 2022-04-20:
Hello there and thank you so much for your great library and work. I want to start playing around with the ULP copro and as a first step, I want to get the blinky example getting to run. I am working with micropython 1.18 on Heltec Wifi Kit32. On this board, the onboard LED is on GPIO25, so I have changed the definition to:

`#define RTCIO_GPIO25_CHANNEL 6`

If I start the program it will run without an error and I still can see the print output changing from 1 to 0 and vice versa, but the LED doesn´t light. Could you give me a hint, of what I am doing wrong?

Thanks in advance and best regards,
Werner

wnienhaus, 2022-04-21:
I had a quick look and I have the feeling the PIN might not be in output mode.

This line sets the output mode: https://github.com/micropython/micropython-esp32-ulp/blob/master/examples/blink.py#L63

Did you change the `RTC_IO_TOUCH_PAD2_REG` and `RTC_IO_TOUCH_PAD2_MUX_SEL_M` appropriately? As per this [mapping table](https://github.com/espressif/esp-idf/blob/4bf67afd0a5a4183f225e0678a38296cbd194291/components/soc/esp32/rtc_io_periph.c#L53), those constants need to be changed to `RTC_IO_PAD_DAC1_REG ` and `RTC_IO_PDAC1_MUX_SEL_M ` respectively (see comments at the end of each line of the table for the "normal" GPIO pin number for reference).

Give it a try. If that doesn't do it - I happen to have a "Heltek Wireless Stick", which also has its onboard LED on GPIO 25 / RTC 6, so I could test it on my board at some point.

wnienhaus, 2022-04-22:
Thanks for letting me know. I will try update the example file to better explain this for future users.