micropython/micropython-esp32-ulp #78: Questiona about GPIO Output (LED) example

wz2b, 2022-07-15:
https://github.com/micropython/micropython-esp32-ulp/blob/8fa707030eb7fc1b87f4140067ba096da7329a41/examples/blink.py#L86

Do I understand in this bit of code that you're turning the led on and off by enabling or disabling the GPIO output?  I'm curious why you would have done that, rather than always initializing the pin as an output, then writing to RTCIO_RTC_GPIO_OUT_REG, RTCIO_RTC_GPIO_OUT_W1TS_REG, or RTCIO_RTC_GPIO_OUT_W1TC_REG?

wnienhaus, 2022-07-21:
You are very correct. That code is indeed incorrect. Some bad copy-pasting that happened to work(ish).

The correct way would be to set the port to output enabled once with this instruction:
```
WRITE_RTC_REG(RTC_GPIO_ENABLE_W1TS_REG, RTC_GPIO_ENABLE_W1TS_S + gpio, 1, 1)
```

and then to toggle the LED on and off with these instructions:
```
//on
WRITE_RTC_REG(RTC_GPIO_OUT_REG, RTC_GPIO_OUT_DATA_S + gpio, 1, 1)

//off
WRITE_RTC_REG(RTC_GPIO_OUT_REG, RTC_GPIO_OUT_DATA_S + gpio, 1, 0)
```

Alternatively one can use the `RTC_GPIO_OUT_W1TS_REG` and `RTC_GPIO_OUT_W1TC_REG` registers, but I prefer the `RTC_GPIO_OUT_REG` approach. I am not sure why there are two ways to do this.

I just tested this with a power profiler (nordic PPK2), which supports reading in logic signals alongside the power profile. When I connect my GPIO output to the logic port with the current (wrong) `blink.py` code, I can see the initial low to high transition, but the logic port never "sees" a low again (I guess with "output enable" turned off the pull-down is also disabled and the pin is left floating).

When I instead fix the code as per above, I correctly see the highs and lows of the GPIO output from via the logic port.

Thanks for picking this up. I will create a PR for this soon.

wz2b, 2022-07-21:
Great, thanks explaining - I _thought_ that was what was going on, but I'm using these examples to learn how to make a virtual SPI port in the ULP and I'm not quite used to the GPIO/RTC muxes in this thing yet.

wnienhaus, 2022-08-19:
@wz2b I have now merged the fix. Thanks again for reporting it.

closed 2022-08-19 by wz2b