micropython/micropython-esp32-ulp #85: Support ESP32-S2

wnienhaus, 2022-09-04:
Support for the S2 should not be that difficult, because the ULP in the ESP32-S2 has only a slightly different binary format than the ULP of the original ESP32.

(Note the S2 has both a RISC based and FSM based ULP - this issue is about adding support for the FSM based ULP)

When implementing this, we should ensure that we have a clean and clear way of selecting which ESP32 variant to use as the target.

rkagerer, 2023-01-29:
Not sure if this belongs in the same issue, but I'd love to see it working on the S3 as well.

wnienhaus, 2023-01-29:
A quick glance at the Technical Reference Manuals for the S2 and S3 suggests that the binary formats of the ULP-FSM are identical across both variants. Thus S2 support should also result in S3 support.

wnienhaus, 2023-06-11:
I have started working on this - already have our counter.py example working, but will need to make my way through all the opcodes to add the new binary format.

Micropython will also need to be built against at least v4.4.2 of the ESP-IDF (currently it builds against 4.4.0) because some important fixes related to the ULP-FSM on S2/S3 have been made in the ESP-IDF since 4.4.0. I will submit a PR to Micropython for that change.

mattytrentini, 2023-06-13:
Worth noting that Damien is planning to move to 5.x. See #[11528](https://github.com/micropython/micropython/pull/11528) for more detail.

Somewhat related - it would be great to test the ULP from a build from the branch on 11528!

wnienhaus, 2023-06-13:
Great. Thanks for the pointer - I had not seen that PR.

I will definitely try the ULP with that branch.

wnienhaus, 2023-06-14:
@mattytrentini I just tested the ULP (FSM) on both the S2 and S3 using the branch of https://github.com/micropython/micropython/pull/11528 and it works perfectly.

The reasons seem to be:
* because the same fixes from v4.4.2 are already included
* and also because espressif seem to have unified/merged the ULP build config flags. In IDF v4.x those configs were per-each-target and so the sdkconfig.base file only enabled the ULP for the original ESP32, rather than for all variants that have a ULP.

So for anyone wanting to test the FSM ULP on the S2/S3 right now, either use Damien's [draft PR for IDFv5.0](https://github.com/micropython/micropython/pull/11528) or my PR which [fixes ULP support for S2/S3](https://github.com/micropython/micropython/pull/11761) against the current master branch.

dpgeorge, 2023-06-14:
Thanks @wnienhaus for testing the IDF 5 branch, that is very helpful.  And I'm glad it works!

PR #91 (2023-09-02): ESP32-S2/S3 support
This PR adds support for the ULP-FSM (not ULP RISC-V) of the ESP32-S2 and ESP32-S3.

(Note, the ESP32-S2 and S3 have the same ULP-FSM, so we don't distinguish between them in the implementation).

The entire instruction set of the ULP-FSM of the ESP32-S2 is supported, including the new load and store instructions to allow loading from or storing to the upper 16-bits of memory locations.

Use the `-c` command line argument when assembling or disassembling. Use `-c esp32` (or omit the `-c` option) to select the original ESP32. Use `-c esp32s2` to select the ESP32-S2/S3. Refer [... cut]
closed 2023-09-02 by wnienhaus