micropython/micropython-esp32-ulp #90: Add License/Copyright to all files

wnienhaus, 2023-07-12:
Currently we have only a general LICENSE file in the repo root.

Having license/copyright information in each file makes it very clear for anyone who copies the file what the license/copyright is.

We can either add a short header using the `SPDX-License-Identifier` format. Or a long one like in the `micropython` repo.

wnienhaus, 2023-07-12:
@ThomasWaldmann commented in the PR (#89):
> I'ld rather dislike having the same (long) license text as a header in each file.
> If there is a header in a file, I'ld rather expect it to roughly tell what's inside that file and after that there could be also a short notice identifying the license.

So perhaps we'll go the `SPDX-License-Identifier` route. Final decision is still open.

dpgeorge, 2023-07-12:
Here's an example of a short header with description, copyright and license: https://github.com/micropython/micropython/blob/master/lib/littlefs/lfs2.c#L1-L7

wnienhaus, 2023-07-13:
That looks pretty neat. One liner description (or longer if needed), followed by (in our case) 1 copyright line, followed by `SPDX-License-Identifier: MIT`. What do you think @ThomasWaldmann ?

e.g.
```py
# ESP32 ULP Co-Processor Assembler
#
# Copyright 2018-2023, the micropython-esp32-ulp authors, see AUTHORS file.
# SPDX-License-Identifier: MIT
```

or maybe:
```py
"""
ESP32 ULP Co-Processor Assembler

Copyright 2018-2023, the micropython-esp32-ulp authors, see AUTHORS file.
SPDX-License-Identifier: MIT
"""
```

even though I wonder whether the 2nd example would be interpreted as a string and use thus more memory?

ThomasWaldmann, 2023-07-13:
If one puts the "current year" there, one has to run around once a year and update the years everywhere.

wnienhaus, 2023-07-14:
Btw, saw in some other code today, there's also a `SPDX-FileCopyrightText` identifier, so we could make it:

```py
# ESP32 ULP Co-Processor Assembler
#
# SPDX-FileCopyrightText: 2018-2023, the micropython-esp32-ulp authors, see AUTHORS file.
# SPDX-License-Identifier: MIT
```

wnienhaus, 2023-09-06:
I have now created PR #95, adding licence headers everywhere.
@ThomasWaldmann @dpgeorge feel free to review and give feedback.

PR #95 (2023-09-11): Add licence headers to all source files
This PR adds a licence header to every source code file to make it very clear for anyone who copies the given file what the licence/copyright is.

Closes #90
closed 2023-09-11 by wnienhaus