micropython/micropython-esp32-ulp #93: Update quickstart to use mip instead of upip

mjaspers2mtu, 2023-08-08:
Micropython now uses mip instead of upip https://docs.micropython.org/en/latest/reference/packages.html#packages

The quickstart guide and the installation method in the [docs](https://github.com/micropython/micropython-esp32-ulp/blob/master/docs/index.rst) do not work on the latest builds.

```py
>>> import upip
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: no module named 'upip'
>>> 
```

The reason I raise this issue is because I can't figure out how to install this ulp repo on my esp32 😅

wnienhaus, 2023-08-08:
I'm working on that one (should have it soon), but currently focusing on finishing S2/S3 support.

For now you could use `mpremote` on a PC to copy the necessary files to your ESP32.

First clone this repo, then from within the repo root, run:

```sh
pip install mpremote  #if not yet installed
mpremote cp -r esp32_ulp :
```

Then it will work.

wnienhaus, 2023-08-08:
Potentially this would work too (I havent tested it):
```py 
# on the ESP32
# make sure network is connected
import mip
mip.install("https://gist.githubusercontent.com/wnienhaus/5a30dd2ac7000764b065b44c5341f1e9/raw/54e1d43dbb0761b10134a34ca70d272064b66987/package.json")
```

mjaspers2mtu, 2023-08-08:
> Potentially this would work too (I havent tested it, but I made the necessary package.json available as a Gist):
> 
> ```python
> # on the ESP32
> # make sure network is connected
> import mip
> mip.install("https://gist.githubusercontent.com/wnienhaus/5a30dd2ac7000764b065b44c5341f1e9/raw/54e1d43dbb0761b10134a34ca70d272064b66987/package.json")
> ```

Wow thanks for the fast reply!

Your command almost worked:
```py
>>> import mip
>>> mip.install("https://gist.githubusercontent.com/wnienhaus/5a30dd2ac7000764b065b44c5341f1e9/raw/54e1d43dbb0761b10134a34ca70d272064b66987/package.json")
Installing https://gist.githubusercontent.com/wnienhaus/5a30dd2ac7000764b065b44c5341f1e9/raw/54e1d43dbb0761b10134a34ca70d272064b66987/package.json to /lib
Copying: /lib/esp32_ulp/__init__.py
Copying: /lib/esp32_ulp/__main__.py
Copying: /lib/esp32_ulp/assemble.py
Copying: /lib/esp32_ulp/definesdb.py
Copying: /lib/esp32_ulp/link.py
Copying: /lib/esp32_ulp/nocomment.py
Copying: /lib/esp32_ulp/opcodes.py
Copying: /lib/esp32_ulp/parse_to_db.py
Copying: /lib/esp32_ulp/preprocess.py
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "mip/__init__.py", line 1, in install
  File "mip/__init__.py", line 1, in _install_package
  File "mip/__init__.py", line 1, in _install_json
  File "mip/__init__.py", line 1, in _download_file
  File "mip/__init__.py", line 1, in _chunk
OSError: -29184
>>> 

```

mjaspers2mtu, 2023-08-08:
> > > Potentially this would work too (I havent tested it, but I made the necessary package.json available as a Gist):
> > > ```python
> > > # on the ESP32
> > > # make sure network is connected
> > > import mip
> > > mip.install("https://gist.githubusercontent.com/wnienhaus/5a30dd2ac7000764b065b44c5341f1e9/raw/54e1d43dbb0761b10134a34ca70d272064b66987/package.json")
> > > ```
> > 
> > 
> > Wow thanks for the fast reply!
> > Your command almost worked:
> > ```python
> > >>> import mip
> > >>> mip.install("https://gist.githubusercontent.com/wnienhaus/5a30dd2ac7000764b065b44c5341f1e9/raw/54e1d43dbb0761b10134a34ca70d272064b66987/package.json")
> > Installing https://gist.githubusercontent.com/wnienhaus/5a30dd2ac7000764b065b44c5341f1e9/raw/54e1d43dbb0761b10134a34ca70d272064b66987/package.json to /lib
> > Copying: /lib/esp32_ulp/__init__.py
> > Copying: /lib/esp32_ulp/__main__.py
> > Copying: /lib/esp32_ulp/assemble.py
> > Copying: /lib/esp32_ulp/definesdb.py
> > Copying: /lib/esp32_ulp/link.py
> > Copying: /lib/esp32_ulp/nocomment.py
> > Copying: /lib/esp32_ulp/opcodes.py
> > Copying: /lib/esp32_ulp/parse_to_db.py
> > Copying: /lib/esp32_ulp/preprocess.py
> > Traceback (most recent call last):
> >   File "<stdin>", line 1, in <module>
> >   File "mip/__init__.py", line 1, in install
> >   File "mip/__init__.py", line 1, in _install_package
> >   File "mip/__init__.py", line 1, in _install_json
> >   File "mip/__init__.py", line 1, in _download_file
> >   File "mip/__init__.py", line 1, in _chunk
> > OSError: -29184
> > >>> 
> > ```

Further after ignoring the oserror, and copying the example file counter.py, I got this:
```py
>>> import counter
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "counter.py", line 15, in <module>
  File "/lib/esp32_ulp/__init__.py", line 1, in <module>
ImportError: no module named 'esp32_ulp.util'
```

I had to copy over util.py, which then results in:

```py
>>> import counter
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "counter.py", line 15, in <module>
  File "/lib/esp32_ulp/__init__.py", line 3, in <module>
ImportError: can't import name preprocess
>>> 
```

wnienhaus, 2023-08-08:
```py
ImportError: can't import name preprocess
```

this will be the same thing as `util.py` - the file did not get installed.

Basically the mip install crashed while downloading `preprocess.py` so it and all files thereafter are corrupt/missing. You can see the full list of files that should have downloaded in the gist: (one more file that should have come after the mip crash during `preprocess.py`).

Make sure to manually copy all the missing files to the `esp32_ulp` directory on your ESP32.

mjaspers2mtu, 2023-08-08:
> ```python
> ImportError: can't import name preprocess
> ```
> 
> this will be the same thing as `util.py` - the file did not get installed correctly.
> 
> Basically the mip install crashed while downloading `preprocess.py` so it and all files thereafter are corrupt/missing. You can see the full list of files that should have downloaded in the gist: (one more file that should have come after the mip crash during `preprocess.py`).
> 
> Make sure to manually copy all the missing files to the `esp32_ulp` directory on your ESP32.

This was exactly the problem thanks a bunch!

wnienhaus, 2023-09-02:
I'll reopen this, because one still cannot install using mip, and the latest MicroPython no longer has upip. (The above is just a workaround).
I'll add the necessary support next, by adding a valid `package.json` file and by updating the documentation.

PR #94 (2023-09-06): Add package.json for MIP package manager
This PR adds a `package.json` file to allow installing `micropython-esp32-ulp` with the `mip` package manager on modern MicroPythons.

To install with mip, one would run:
```py
import mip
mip.install('github:micropython/micropython-esp32-ulp')

# note: while this PR is not merged, this command can be used for testing instead
mip.install('https://raw.githubusercontent.com/wnienhaus/micropython-esp32-ulp/mip_support/package.json')
```

This PR also adds a tool for generating the contents of `package.json`, so it doesn't have to be handcrafted whenever fil [... cut]
closed 2023-09-06 by wnienhaus