micropython/micropython-esp32-ulp #99: binary file should be open with 'rb' instead of 'r' in advanced exemple

pidou46, 2024-11-03:
With mp 1.24 , in advanced usage exemple: <with open('test.ulp', 'r') as f:> rise a unicode exception

It should be changed to:
<with open('test.ulp', 'rb') as f:>

ThomasWaldmann, 2024-11-03:
guess this refers to item 3 there: https://github.com/micropython/micropython-esp32-ulp/blob/master/docs/index.rst#advanced-usage

sure, reading binary files should always use "rb" mode.

can you submit a PR?

pidou46, 2024-11-03:
Yes, it's.
  
I will try to submit a PR, m'y knowlage with git is very thinght but the modification is trivial so I shoud be able to do it.
  
The code worked with earlier mp version, so I guess I should try to see if this PR would work with previous mp versions (1.12 ?) priori to submit it?

ThomasWaldmann, 2024-11-03:
I don't think you need to check with old micropython versions, just with the current one.

wnienhaus, 2024-12-15:
Fixed by #100

closed 2024-12-15 by wnienhaus