micropython/micropython-esp32-ulp #103: how to have more variables in ULP code

vencax, 2024-12-19:
Hi, i have this code:
```
...
counter:	.long 0

entry:
...            
```
but when I add mode variable, it stop working:
```
...
counter:	.long 0
threshold:      .long 0

entry:
...       
```

what I am doing wrong?

vencax, 2024-12-20:
Ha, silly bug. Its necessary to set proper __entry_addr__ when starting the ULP program. When there is 2 variables, then __entry_addr__ must be 8. Following formula can be used:
```
entry_addr = 4 * <nuber of variables>
```
because variables are aligned in memory to 4 bytes.

NOTE:
I was struggling with sections in asm as well. I tried put variables into .data or into .bss sections. Result:
__there is no need to use sections. You can omit sections completely (the code is put into .text section by default).__

closed 2024-12-20 by vencax

wnienhaus, 2024-12-21:
Great. That is what I would have assumed - about the entry point offset.
However, are you saying that using the .data or .bss sections has no impact?
Theoretically anything in .data should go after the code (.text) and anything from .bss goes after data (without actually being put into the image).